riscv-software-src/riscof #1: support for intermediate run start points

neelgala, 2021-07-16:
Currently the `riscof run` command starts always by creating a database of the suite, generating a test-list from it, and then running the tests. However the database generation and test-list generation are quite time consuming functions. While debugging bugs one would like to skip these everytime and simply start from the test-list itself. 

Proposal:
To the `run` command add 2 more sub args: `--dbfile` and `--testlist`. These arguments along with the `--suite` will all be mutually exclusive. Following is what is done for each args:

- `--suite`: run database generation, generate test-list and run tests on target and reference
- `--dbfile`: generate test-list and run tests on target and reference (assumes the dbfile is correct and updated)
- `--testfile`: directly run tests on the target and reference (assumes the testlist has been correctly generated)

Only one of the above should be enabled at any point.

One other subargs could be to run the tests only on the reference or the target and then quit (before performing the signature comparisons) This can be achieved by having `--no-ref-run` and `--no-dut-run` sub args to the `run` command

bilalsakhawat-10xe, 2021-07-19:
Hi @neelgala! Can you assign this issue to me?

neelgala, 2021-07-19:
@bilalsakhawat-10xe assigned to you. I would also suggest branching off from migration-fix branch.. that is a soon to be merged branch to master with some pretty large changes.

bilalsakhawat-10xe, 2021-07-19:
Thanks @neelgala. Sure I'll branch off from migration-fix branch.

neelgala, 2021-08-03:
fixed in #12

closed 2021-08-03 by neelgala