riscv-software-src/riscof #10: Connect Riscof to SweRV EH2 core

SyoAnd, 2021-07-23:
How would I use riscof to validate the SweRV RH2 core from Western Digital?
https://github.com/chipsalliance/Cores-SweRV-EH2

I expect that I have to use an EDA tool, like Cadence Xcelium, but I have no idea how to connect riscof to it.
Maybe I am blind, but the documentation mentions the ability to validate RTL code many times but does never state how to do it.

Currently, I have a working Riscof setup with spike and sail_model.

pawks, 2021-07-23:
Anything which can be run via the command line and doesn't need a GUI to be used can be used with riscof. The plugins for SAIL and SPIKE write out Makefiles at `riscof-work/Makefile.<name>` which are then used to actually run the tests on the core and generate signatures. Something similar can be done for any RTL/simulator.

The readme in the SweRV repository shows that there is support for generating a verilator simulator of the core. Verilator is a free to use tool and can generate a binary for the RTL, which can be used to run programs. You will have to write a custom plugin for the same. [This](https://riscof.readthedocs.io/en/stable/plugins.html) guide documents the architecture of the plugins and what each function in the plugin should do. You can find a few example plugins [here](https://gitlab.com/incoresemi/riscof-plugins/-/tree/master). You will also need to define the DUT specific macros in a custom header file and a custom linker script based on the RTL. The plugin for the [cClass core from SHAKTI, IITM](https://gitlab.com/shaktiproject/cores/c-class/-/tree/master) can be found [here](https://gitlab.com/incoresemi/riscof-plugins/-/tree/stale-plugins/cClass) and can be used for inspiration. Please note that the plugin was for an older version of riscof and has not been updated to accommodate the recent changes. Hence it might not work with the latest version of riscof.

neelgala, 2021-07-23:
You can also refer to the chromite-core plugin [here](https://gitlab.com/incoresemi/riscof-plugins/-/tree/master/chromite) which should work with the latest riscof version. It has a readme and the plugin files are also well commented. It should provide a decent starting point.

haseebazaz-10xe, 2021-08-13:
I am integrating SweRV-el2 to run with riscof environment taking chromite as reference. I created a directory in riscof-plugins with all the files creating according to SweRV-el2 like config.ini, env, isa, platform and python file too and in core I was also able to create executable but when run core with riscof environment I get error that some memory file is missing. I also looked into chromite and see that there is boot.mem file too which I don't see in SweRV directory and I don't know how to generate boot.mem file in case of SweRV. As in case of chromite there is .dts file which is used to generate boot.mem but I don't see any in case of SweRV. What should I do now?

pawks, 2021-08-13:
> I am integrating SweRV-el2 to run with riscof environment taking chromite as reference. I created a directory in riscof-plugins with all the files creating according to SweRV-el2 like config.ini, [...]

The SweRV repo might be a better to place to ask this question. I had a quick glance over at the [makefile](https://github.com/chipsalliance/Cores-SweRV-EH2/blob/a95fdb81ea6dc19239cfe2953ef00710a6f1cf2a/tools/Makefile#L154) in the repository and looks like the SweRV core does not require a `boot.mem` file. The compiled binary can be directly executed on the core.