riscv-software-src/riscof #13: Add hartid argument to cli.

pawks, 2021-08-12:
It is currently assumed that the hartid being tested is `0`. This should be an argument in the cli to enable testing harts with other hartid values.

anonymousr007, 2022-01-18:
Hello @pawks, I am Rishabh, an undergraduate student in computer science in India. Can you elaborate a little bit more? I would like to make a pull request for this issue 😄

pawks, 2022-01-18:
The configuration for the hart is taken from the config yaml supplied by the plugins in the riscv-config format. The format supports configurations for multiple cores(harts) like in the example [here](https://github.com/riscv-software-src/riscof/issues/31#issuecomment-1006481828). Currently it is assumed that the hart being tested is `hart0`. This feature would allow the hard id to be specified from the cli.