riscv-software-src/riscof #16: Missing dependency: `device-tree-compiler`

kaddkaka, 2021-08-30:
I'm following the [installation guide](https://riscof.readthedocs.io/en/stable/installation.html), but there seems to be dependencies missing form the instructions.

The step `apt-get install device-tree-compiler` as mentioned at https://github.com/riscv/riscv-isa-sim#build-steps is missing.

kaddkaka, 2021-08-30:
Also, I couldn't not build spike with clang12 (got warnings in boost source code). But building with gcc worked fine.

Example error from clang build:

```
In file included from ../riscv/sim.cc:3:
In file included from ../riscv/sim.h:9:
In file included from /usr/include/boost/algorithm/string.hpp:19:
In file included from /usr/include/boost/algorithm/string/trim.hpp:19:
In file included from /usr/include/boost/range/as_literal.hpp:22:
In file included from /usr/include/boost/range/iterator_range.hpp:13:
In file included from /usr/include/boost/range/iterator_range_core.hpp:27:
In file included from /usr/include/boost/iterator/iterator_facade.hpp:14:
In file included from /usr/include/boost/iterator/iterator_categories.hpp:17:
In file included from /usr/include/boost/mpl/placeholders.hpp:24:
In file included from /usr/include/boost/mpl/arg.hpp:25:
In file included from /usr/include/boost/mpl/aux_/na_assert.hpp:23:
/usr/include/boost/mpl/assert.hpp:174:20: error: expected a qualified name after 'typename'
template< typename P > struct assert_arg_pred
                   ^
/sw/riscv/riscv-isa-sim/build/../riscv/decode.h:177:11: note: expanded from macro 'P'
#define P (*p)
```

neelgala, 2021-08-30:
I have pushed a fix for the docs on [this](https://github.com/riscv/riscof/tree/plugin-doc-update) branch. Will be merging soon. 

Regarding spike with clang build,  I guess that is a question best posted at the riscv-isa-sim repository.

neelgala, 2021-09-10:
new docs fix this.. hence closing this.

closed 2021-09-10 by neelgala