riscv-software-src/riscof #17: riscof validateyaml: ruamel yaml reader choking on special character

kaddkaka, 2021-08-30:
When trying to `riscof validateyaml --config=config.ini`([here](https://riscof.readthedocs.io/en/stable/installation.html#running-riscof)) with the template ini file generated by `riscof setup --dutname=spike` ([here](https://riscof.readthedocs.io/en/stable/installation.html#create-neccesary-env-files)) I get the following error:

```
     INFO | [--riscv_config.checker--]: Load Schema /home/bo/.local/lib/python3.6/site-packages/riscv_config/schemas/schema_isa.yaml
Traceback (most recent call last):
  File "/home/bo/.local/bin/riscof", line 11, in <module>
    sys.exit(execute())
  File "/home/bo/.local/lib/python3.6/site-packages/riscof/main.py", line 211, in execute
    isa_file = checker.check_isa_specs( isa_file, work_dir, True)

   [...]

  File "/home/bo/.local/lib/python3.6/site-packages/ruamel/yaml/reader.py", line 249, in check_printable
    'special characters are not allowed',
ruamel.yaml.reader.ReaderError: unacceptable character #x0080: special characters are not allowed
  in "/home/bo/.local/lib/python3.6/site-packages/riscv_config/schemas/schema_isa.yaml", position 15493
```

This corresponds to the `’` at `~/.local/lib/python3.6/site-packages/riscv_config/schemas/schema_isa.yaml:493`:
`           default: The mstatus register keeps track of and controls the hart’s current`

My installed python version is:
```
> python3 --version
Python 3.6.9
```

neelgala, 2021-08-30:
what version of riscv-config are you using ? `riscv-config --version`

kaddkaka, 2021-09-06:
```
> riscv-config --version
RISCV-CONFIG: RISC-V Configuration Validator
Version: 2.10.1
```

neelgala, 2021-09-10:
well.. I am not able to replicate this error at my end ? Could try a fresh run with the latest riscof (pip install -U riscof) maybe and see if you still see this issue ?