riscv-software-src/riscof #18: Quickstart: riscof testlist: argument requirements unclear

kaddkaka, 2021-08-30:
According to https://riscof.readthedocs.io/en/stable/installation.html#running-riscof `riscof testlist` doesn't need the arguments `--suite` and `--env` but this seems to be incorrect since the command `riscof testlist --config=config.ini` results in this output:

```
error: the following arguments are required: --suite, --env
usage: riscof testlist [-h] [--work-dir PATH] [--config PATH] --suite PATH
                       --env PATH

optional arguments:
  --config PATH    The Path to the config file. [Default=./config.ini]
  --env PATH       The Path to the custom env directory.
  --suite PATH     The Path to the custom suite directory.
  --work-dir PATH  The Path to the work-dir.
  -h, --help       show this help message and exit
```

But as you see the help text is a bit of a mix, both listing them as required *and* optional.

neelgala, 2021-08-30:
@kaddkaka Yes the quickstart is a little dated. We are working improving the docs on [this](https://github.com/riscv/riscof/tree/plugin-doc-update) branch. Will have them updated and merge soon.

Also the help-command doesn't yet distinguish between required and non-required arguments. I think we will be moving to click based cli to ensuer a better help screen.

neelgala, 2021-08-30:
https://riscof.readthedocs.io/en/plugin-doc-update/installation.html#quickstart following this should solve your problems

kaddkaka, 2021-09-07:
Thanks @neelgala this updated documentation suggests this command:

`riscof testlist --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env`

But what path is actually `--suite=riscv-arch-test/riscv-test-suite/` ? For example for the spike template, what should the command look like?

`riscof testlist --config=config.ini --suite=??? --env=spike/env`

neelgala, 2021-09-10:
the branch with docs has been merged . Closing this issue.

closed 2021-09-10 by neelgala