riscv-software-src/riscof #23: Swerv EH1 Core Compliance Test

YeuzhiHere, 2021-09-24:
I want to do compliance test for Swerv EH1 Core using riscof. But I got following error after installing riscof and running "riscof --help"
[image]

Please guide and resolve this issue.

regards

pawks, 2021-09-24:
Could you please post the details of the environment you are using and also the output of the `pip list` command.

YeuzhiHere, 2021-09-27:
Yeah, I am using Ubuntu 20.04. And output for pip list is;
[image]
[image]
[image]
[image]

pawks, 2021-09-27:
Upgrading the version of `click` to the latest should fix your error. Try the following command:
`pip install -U click`

YeuzhiHere, 2021-09-27:
Ok

closed 2021-09-27 by YeuzhiHere