riscv-software-src/riscof #29: Quickstart SPIKE DUT fails with AttributeError: 'Token' object has no attribute 'test'

Kaucasus, 2021-11-28:
When trying to follow the quickstart guide as in: [https://riscof.readthedocs.io/en/stable/installation.html#](https://riscof.readthedocs.io/en/stable/installation.html#) the `riscof run` command at the very end fails.

I've followed the installation step by step on a clean Ubuntu 20.04 LTS install in Windows Subsystem for Linux with the following directories and decisions made with every step succeeding: 

1. Installed and used pyenv for managing my Python versions in `~/.pyenv` and activated the shell
2. Installed RISCOF via git
3. Installed the RISCV-GNU toolchain with ./configure install prefix path `~/riscof`
4. Cloned Spike in `~/riscv-sims/riscv-isa-sim` and ./configure install prefix path `~/riscv-sims/spike`
5. Cloned Sail in `~/riscv-sims/sail-riscv`
6. In folder `~/risctest/spike`, ran the `riscof setup --dutname=spike` command
7. Changed the ini file to include the path to riscv_sim_RV32
8. Cloned the architectural tests without problems, and validated the yaml and generated the test list without problem

Then, finally running in the same directory `riscof run --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env` fails with a Python AttributeError: 'Token' object has no attribute 'test'.

It does not open up a HTML page with the test results, and just exits.
Seemingly, the tool does simulate and collects the test signature for both the reference and the dut. The tool also seemingly runs perfect up to that point, and logs the info messages to the shell no problem.
A full stacktrace can be found below:
```shell
user@COMPUTER:~/risctest/spike$ riscof run --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.23.2 *******
    INFO | using riscv_isac version : 0.8.0
    INFO | using riscv_config version : 2.11.1
    INFO | Reading configuration from: /home/user/risctest/spike/config.ini
    INFO | Preparing Models
    INFO | Input-ISA file
    INFO | Loading input file: /home/user/risctest/spike/spike/spike_isa.yaml
    INFO | Load Schema /home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/riscv_config/schemas/schema_isa.yaml
    INFO | Processing Hart: hart0
    INFO | Initiating Validation
    INFO | No errors for Hart: 0 :)
    INFO | Initiating post processing and reset value checks.
    INFO | Dumping out Normalized Checked YAML: /home/user/risctest/spike/riscof_work/spike_isa_checked.yaml
    INFO | Input-Platform file
    INFO | Loading input file: /home/user/risctest/spike/spike/spike_platform.yaml
    INFO | Load Schema /home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/riscv_config/schemas/schema_platform.yaml
    INFO | Initiating Validation
    INFO | No Syntax errors in Input Platform Yaml. :)
    INFO | Dumping out Normalized Checked YAML: /home/user/risctest/spike/riscof_work/spike_platform_checked.yaml
    INFO | Generating database for suite: /home/user/risctest/spike/riscv-arch-test/riscv-test-suite
    INFO | Database File Generated: /home/user/risctest/spike/riscof_work/database.yaml
    INFO | Env path set to/home/user/risctest/spike/riscv-arch-test/riscv-test-suite/env
    INFO | Running Build for DUT
    INFO | Running Build for Reference
    INFO | Selecting Tests.
    INFO | Running Tests on DUT.
    INFO | Running Tests on Reference Model.
    INFO | Initiating signature checking.
    INFO | Following 90 tests have been run :

    INFO | TEST NAME                                          : COMMIT ID                                : STATUS
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cadd-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/caddi-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/caddi16sp-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/caddi4spn-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cand-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/candi-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cbeqz-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cbnez-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cebreak-01.S : 0e77784916ed9c07842883ba6c62db2555a8335f : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cj-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cjal-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cjalr-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cjr-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cli-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/clui-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/clw-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/clwsp-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cmv-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cnop-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cor-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
    INFO | /home/user/risctest/spike/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cslli-01.S : 6ba8712973dade81eaf1182d1050e5799a7ef2a0 : Passed
[... 88 lines cut ...]
    return cls.from_code(self, self.compile(source), gs, None)
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/environment.py", line 749, in compile
    source = self._parse(source, name, filename)
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/environment.py", line 606, in _parse
    return Parser(self, source, name, filename).parse()
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 1038, in parse
    result = nodes.Template(self.subparse(), lineno=1)
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 1012, in subparse
    add_data(self.parse_tuple(with_condexpr=True))
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 725, in parse_tuple
    args.append(parse())
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 523, in parse_expression
    return self.parse_condexpr()
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 528, in parse_condexpr
    expr1 = self.parse_or()
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 543, in parse_or
    left = self.parse_and()
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 552, in parse_and
    left = self.parse_not()
  File "/home/user/.pyenv/versions/3.6.0/lib/python3.6/site-packages/jinja2/parser.py", line 560, in parse_not
    if self.stream.current.test("name:not"):
AttributeError: 'Token' object has no attribute 'test'
```

If anybody has an idea how to solve this issue, or if you need more information about my file structure, do let me know!

neelgala, 2021-11-28:
@Kaucasus thanks for reporting this. Can I ask what python version are you using ? Also what version of riscof you are using (riscof --version should do the trick)

@pawks any thoughts ?

Kaucasus, 2021-11-28:
The Python version as reported by `Python --version` is 3.6.0 (managed by pyenv), and riscof is `RISC-V Architectural Test Framework., version 1.23.2`

pawks, 2021-11-28:
As far as I could trace it out, it seems to be an error due to using `python 3.6.0` and `Jinja 3.0.1`. Please try downgrading `Jinja` to 3.0.0 using `pip install jinja==3.0.0` or upgrade to python `3.6.x`. 

Note: If you are using `pyenv` it is better to shift to one of the more recent stable versions of python like `3.7.x` or later.

Kaucasus, 2021-11-28:
I updated my python environment via pyenv to target 3.8.10. Installing riscof in that environment works as intended!
(well, technically because I run WSL the browser can't automatically open, but the report.html gets generated, and running with the `--no-browser` flag exists cleanly).

Thank you very much for the quick and speedy resulotion!

neelgala, 2021-11-28:
@pawks since riscof is meant for python 3.6 maybe we should freeze Jinja at 3.0.0 in the requirements.txt ?

pawks, 2021-11-28:
That would be a short term fix for sure. However, official support for python 3.6 ends in December 2021 and many problems may arise in future when other packages also start causing issues.

pawks, 2022-04-06:
Tracking the issue in #43

closed 2022-04-06 by pawks