riscv-software-src/riscof #30: Some arch tests are failing with riscof

owaisfarooq-10x, 2021-12-09:
I followed the `riscof` installation quickstart guide and when I ran the tests using the following command some tests passed and others failed

`riscof run --config=config.ini --suite /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m --env /home/owaisfarooq/riscv-arch-test/riscv-test-suite/env/`

riscof version: RISC-V Architectural Test Framework., version 1.23.2
python3 version: 3.8.10 

```
    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.23.2 *******
    INFO | using riscv_isac version : 0.8.0
    INFO | using riscv_config version : 2.10.1
    INFO | Reading configuration from: /home/owaisfarooq/riscof_test/config.ini
    INFO | Preparing Models
    INFO | Input-ISA file
    INFO | Loading input file: /home/owaisfarooq/riscof-plugins/spike_simple/spike_simple_isa.yaml
    INFO | Load Schema /home/owaisfarooq/.local/lib/python3.8/site-packages/riscv_config/schemas/schema_isa.yaml
    INFO | Processing Hart: hart0
    INFO | Initiating Validation
    INFO | No errors for Hart: 0 :)
    INFO | Initiating post processing and reset value checks.
    INFO | Dumping out Normalized Checked YAML: /home/owaisfarooq/riscof_test/riscof_work/spike_simple_isa_checked.yaml
    INFO | Input-Platform file
    INFO | Loading input file: /home/owaisfarooq/riscof-plugins/spike_simple/spike_simple_platform.yaml
    INFO | Load Schema /home/owaisfarooq/.local/lib/python3.8/site-packages/riscv_config/schemas/schema_platform.yaml
    INFO | Initiating Validation
    INFO | No Syntax errors in Input Platform Yaml. :)
    INFO | Dumping out Normalized Checked YAML: /home/owaisfarooq/riscof_test/riscof_work/spike_simple_platform_checked.yaml
    INFO | Generating database for suite: /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m
    INFO | Database File Generated: /home/owaisfarooq/riscof_test/riscof_work/database.yaml
    INFO | Env path set to/home/owaisfarooq/riscv-arch-test/riscv-test-suite/env
    INFO | Running Build for DUT
    INFO | Running Build for Reference
    INFO | Selecting Tests.
    INFO | Running Tests on DUT.
    INFO | Running Tests on Reference Model.
    INFO | Initiating signature checking.
    INFO | Following 116 tests have been run :

    INFO | TEST NAME                                          : COMMIT ID                                : STATUS
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cadd-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/caddi-01.S : -                                        : Passed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/caddi16sp-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/caddi4spn-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/caddiw-01.S : -                                        : Passed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/caddw-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cand-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/candi-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cbeqz-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cbnez-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cebreak-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cj-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cjalr-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cjr-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cld-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cldsp-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cli-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/clui-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/clw-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/clwsp-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cmv-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cnop-01.S : -                                        : Passed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cor-01.S : -                                        : Passed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csd-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csdsp-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/cslli-01.S : -                                        : Passed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csrai-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csrli-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csub-01.S : -                                        : Failed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csubw-01.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/C/src/csw-01.S : -                                        : Failed
[... 64 lines cut ...]
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/M/src/remw-01.S : -                                        : Passed
    INFO | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/Zifencei/src/Fencei.S : -                                        : Passed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/ebreak.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/ecall.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-beq-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-bge-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-bgeu-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-blt-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-bltu-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-bne-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-jal-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-ld-01.S : -                                        : Failed
[... cut ...]
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign-sw-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign1-jalr-01.S : -                                        : Failed
   ERROR | /home/owaisfarooq/riscv-arch-test/riscv-test-suite/rv64i_m/privilege/src/misalign2-jalr-01.S : -                                        : Failed
    INFO | Test report generated at /home/owaisfarooq/riscof_test/riscof_work/report.html.
    INFO | Opening test report in web-browser

```

neelgala, 2021-12-09:
couple of questions:
1. What version of spike and sail are you using? Are they the latest versions?
2. can you share the report.html file here ?

pawks, 2021-12-10:
@owaisfarooq-10x I tried to recreate this with the latest version of riscv-toolchain, SPIKE, SAIL and riscof with a fresh install. All the tests pass. Please check the versions of the tools you are using and share the `report.html` file.

owaisfarooq-10x, 2021-12-10:
I used the latest version of every tool. I also used the latest sail plugins available at riscof-plugins [repo](https://gitlab.com/incoresemi/riscof-plugins). The [issue](https://gitlab.com/incoresemi/riscof-plugins/-/issues/18) is with riscof-plugins for sail.