riscv-software-src/riscof #31: [Question] How important is CSR configuration?

Don-Haugaard, 2022-01-06:
Hi,

I am currently applying Riscof on a core compiled with Verilator. I have found inspiration from [https://github.com/riscv-software-src/riscof/issues/10](https://github.com/riscv-software-src/riscof/issues/10), which in general has helped me a lot. Thank you for the extended answer in this issue.

In the above issue it is written "_Currently test selection is only based on the ISA of the core for the most part_ (...)"? Furthermore, in the quickstart guide and chromite example, the only configured CSR is misa. So, my question is, if any of you can elaborate on this line? Will it effect my test results if I leave out standard CSR's which have been configured, custom CSR's and other harts from the isa-yaml file? Or is this "only" important if you are using Spike as a DUT?

I hope, I make sense :)

Regards
Don-Haugaard

pawks, 2022-01-06:
Currently the tests available in the `arch-test` repository are catered to only the unprivileged ISA(with the exception of a few tests in the `privilege` directory. But these tests only test whether the trapping behaviour in certain scenarios are correct. They are not configurable based on the behaviour of the CSRs). Hence the test selection happens only based on the ISA string. Future test selection and behaviour will depend on the configuration of the registers. 

To answer your question: No test results are affected if you leave out the standard CSR descriptions as of now. But they will affect the results in future.

Don-Haugaard, 2022-01-06:
O.k, I see - thank you!

Before I close this issue, can I ask you if you have any examples of a isa-yaml that contains more than one hart? My current configuration is accepted by the tool, but I would like to compare it against an example.

neelgala, 2022-01-06:
an example would be something like below. We have 3 harts with hart-ids : 0, 1, 2. hart-0 is RV64, hart-1 is RV32 and hart-2 is the exact same config of hart-0.
```
hart_ids: [0, 1, 2]
hart0: &hart0
    ISA: RV64IMAFDCSUZicsr_Zifencei
    User_Spec_Version: '2.3'
    pmp_granularity: 1
    physical_addr_sz: 32
    supported_xlen: [64]
hart1:
    ISA: RV32IMACUZicsr_Zifencei
    User_Spec_Version: '2.3'
    pmp_granularity: 1
    physical_addr_sz: 32
    supported_xlen: [32]  
hart2: *hart0
```

Don-Haugaard, 2022-01-06:
Thank you very much!

closed 2022-01-06 by Don-Haugaard

neelgala, 2022-01-06:
@Don-Haugaard  btw..once you manage to port your DUT on riscof and if those riscof files are in public-domain.. I would request you to populate the PLUGINS.rst file in this repo with your link ?