riscv-software-src/riscof #35: Mechanism needed to append to test RVTEST_ISA string

nadiga-ventana, 2022-02-23:
All riscv tests have RVTEST_ISA macro fro which the isa string is extracted to be passed as -march option to the compiler.
The rv64i_m/C/src/cebreak-01.S has RVTEST_ISA("RV64IC")
but it also defines rvtest_mtrap_routine which includes CSRRW instruction. Our new compiler (riscv64-unknown-elf-gcc) needs the -march option to have Zicsr to recognize these csrrw instructions.
Is there a way to append to the RVTEST_ISA string ?

pawks, 2022-02-23:
You will need to add `_Zicsr` in the argument to `RVTEST_ISA` for all the tests which need this extension. You will also need to modify the plugin to include the extension in the output `march` argument if it is present in the string.

allenjbaum, 2022-02-23:
I had the same thought. The tests have no visibility as to whether any of the RVMODEL_* macros use CSRs or not - not just RVMODEL_BOOT. It wouldn't be surprising if RVMODEL_HALT did, for example.
Therefore we either need to have all tests default to Zicsr, or require the vendors to explicitly set it ...somewhere... if they use CSRs in their support macros. Their config YAML should have this information

pdonahue-ventana, 2022-02-23:
> Therefore we either need to have all tests default to Zicsr, or require the vendors to explicitly set it ...somewhere... if they use CSRs in their support macros. Their config YAML should have this information

I agree that the -march string should be based on the config YAML instead of RVTEST_ISA.  RVTEST_ISA should only affect whether or not the test runs at all.  The full list of extensions listed in the YAML ought to be available to RVMODEL_* macros so somebody can, for instance, initialize whatever state they need to initialize even if that state uses extensions that are outside of the minimal set used by the test itself.

pawks, 2022-02-23:
The objective of the `RVTEST_ISA` macro is to convey the minimal list of extensions needed for the test to be compiled. This is to serve as a hint for the plugins to derive the `march` string. If any extra extensions are needed for any of the `RV_MODEL*` macros, the plugin is free to add them to the `march` string(care should be taken that any conflicting extensions are not included, as far as the current tests go the C extension shouldn't be add unless specified). The vendors have control over how the `march` is derived from the ISA mentioned in the test. The selection of the test is not dependent on the `RVTEST_ISA` macro, it is dependent on the conditions listed in the `RVTEST_CASE` macro.

allenjbaum, 2022-02-23:
OK, then we just need to document very clearly (but mostly very
prominently) somewhere that
Zicsr must be added to the march string (if not already added ) if
RVMODEL_ macros have any CSR reads or writes.
I'm sure we will get people filing issues because they missed that bit of
information.