riscv-software-src/riscof #40: how to use riscof with verilator

lwshowl, 2022-03-13:
I have read the custom plugins of the document
but still not figure out how to use riscof with verilator
where verilator does not takes binary files

should i create a custom riscof Makefile and modify the run command ?
i also haven't figured out where to create the makefile >_>

sorry for being dumb

pawks, 2022-03-13:
@lwshowl I assume you have a mechanism of loading the program onto your DUT at runtime and that you use the gnu toolchain to compile the tests.  In such a scenario, you will have verilated the model and compiled it into a C binary before running riscof. In the plugin for your DUT you will have to modify line 46 from [here](https://riscof.readthedocs.io/en/stable/plugins.html#makefile-flow-recommended) or line 51 [here](https://riscof.readthedocs.io/en/stable/plugins.html#using-shell-commands) to execute the test on your DUT and get the signature into the appropriate file. [This](https://gitlab.com/incoresemi/riscof-plugins/-/tree/master/chromite) plugin for the chromite core and [this](https://gitlab.com/incoresemi/riscof-plugins/-/tree/stale-plugins/cClass) plugin for the cClass core from Shakti both use verilator for simulation and can be used for inspiration. Note: The cClass plugin is outdated and might not work with the latest version of riscof.

The makefile is created by the plugin and the lines which I spoke about earlier are the lines you will need to modify in your plugin to change the run command.

lwshowl, 2022-03-14:
@pawks thanks for the detailed explanation
i think i have got the trick

lwshowl, 2022-03-14:
@pawks another question here , since the execution environment is occupied by the test program , how can i make a memory dump and make use of the defined macros ?
is the .tohost section in the linker script reserved for the host to dump memory ?

pawks, 2022-03-14:
The DUT specific macros can be defined by you in a header file named `riscv_model.h` and it gets included in the tests during compilation. The linker script should also be defined by you. Only the sections being defined in the test are reserved. You can declare any additional sections you need as necessary and include them appropriately. At the start of the test, the code under RVMODEL_BOOT is executed where you can execute model specific initialisation code. Similarly you have RVMODEL_HALT which defines the HALT sequence for your model. You can have a custom sequence here which sequentially reads the data in the signature region and dumps the same. More reference regarding the model specific macros and what they should do can be found [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#432-required-model-defined-macros).