riscv-software-src/riscof #53: Documentation: inconsistent handling of `config.ini` variable `target_run`

jeras, 2022-06-28:
The documentation in for `target_run` uses inconsistent formatting for the 0 and 1 values.
https://github.com/riscv-software-src/riscof/blame/master/docs/source/plugins.rst#L206-L211

Further from the Python code it is unclear whether this is supposed to be a string or an integer.
But this might just be my lack of understanding of how `ini` files are handled in Python.
https://github.com/riscv-software-src/riscof/blame/master/docs/source/plugins.rst#L259

pawks, 2022-06-28:
In the `ini` files, all values are interpreted as strings while parsing. To compare them as integers in the python code they will need to be converted before use.

jeras, 2022-06-28:
Thanks, I know how to handle this in the documentation change now.