riscv-software-src/riscof #62: D extension must have F and D as part of ISA string

ptprasanna, 2022-09-08:
https://github.com/riscv-software-src/riscof/blob/a003752ff95dd5f9153c323dc2eb8ff7a0426dd9/riscof/framework/main.py#L132

As per the spec "The D extension depends on the base single-precision instruction subset F", which requires both F and D to be passed as part of the ISA string in-order to execute the double precision tests. So when we run a case with `flen = 64`, the current code always passed it as 32 and the statement under `elif` won't get executed at all for the `D` extension tests.

neelgala, 2022-09-08:
the `elif` should just be an `if`. please raise a PR with the fix.

ptprasanna, 2022-09-08:
@neelgala It's done [here](https://github.com/riscv-software-src/riscof/pull/63).

ptprasanna, 2022-09-12:
This is fixed with this [merge](https://github.com/riscv-software-src/riscof/pull/63).

closed 2022-09-12 by ptprasanna