riscv-software-src/riscof #79: where to set supported_xlen to 64

hyperion009, 2023-02-14:
XLEN's default setting is 64.  Where to set supported_xlen to 64? I got thi following issue:
ERROR | riscv32-unknown-elf-objdump: executable not found. Please check environment setup

I am intended to run test with 64 and only settup the   riscv64-unknown-elf-xxx

pawks, 2023-02-14:
You will have to specify the correct isa of your implementation in your isa configuration yaml. Incase you have only implemented the riscv64-* toolchain but you are testing for a RV32 implementation, you can just rename the tool binary appropriately or change the plugin to use the correct toolchain.

xeasonx, 2023-11-08:
Make sure RV64 toolchain and `riscv_sim_RV64` are in PATH.  Edit `spike_isa.yaml` under `spike` dir, change `supported_xlen` to `[64]`, change field `rv32` to `rv64`, then run the test.  Hope help.
```yaml
hart_ids: [0]
hart0:
  ISA: RV64IMCZicsr_Zifencei
  physical_addr_sz: 32
  User_Spec_Version: '2.3'
  supported_xlen: [64]
  misa:
   reset-val: 0x40001104
   rv64:
     accessible: true
     mxl:
       implemented: true
       type:
           warl:
              dependency_fields: []
              legal:
                - mxl[1:0] in [0x1]
              wr_illegal:
                - Unchanged
     extensions:
       implemented: true
       type:
           warl:
              dependency_fields: []
              legal:
                - extensions[25:0] bitmask [0x0001104, 0x0000000]
              wr_illegal:
                - Unchanged
```

Vinayakamk, 2024-03-07:
the above dicussion was helpfull,
i did the same as @xeasonx said, but for me it is still showing the same error.
[image]

i wish to run on ```riscv64-unknown-elf-gcc```