riscv-software-src/riscof #91: truncated isa string

davidharrishmc, 2023-05-09:
I'm getting errors from riscof due to a truncated isa string.  This happens on one machine but not another and I'm not certain of the difference.  Both are Ubuntu 20.04.6 LTS, GCC 12.2.0.

riscof generates a bad Makefile for various tests such as cebreak (below).  Notice that -march=rv32ic_.  On one machine, gcc gives an "error unrecognized opcode csrrw..., exetension zicsr required" message.  When I invoke the command manually with -march=rv32ic_zicsr, it works.  On the other computer, I cebreak seems to compile but I get errors later on.

My spike/spike_rv32gc_isa.yaml contains

  ISA: RV32IMAFDCZicsr_Zifencei_Zba_Zbb_Zbc_Zbs

Why is riscof producing the truncated -march=rv32ic_ string?  Is there a known fix?

Thank you,

David

.PHONY : TARGET40
TARGET40 :
	@cd /home/harris/cvw/tests/riscof/riscof_work/rv32i_m/C/src/cebreak-01.S/dut; riscv64-unknown-elf-gcc -march=rv32ic_           -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -g         -T /home/harris/cvw/tests/riscof/spike/env/link.ld         -I /home/harris/cvw/tests/riscof/spike/env/         -I /home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/env /home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cebreak-01.S -o my.elf  -Drvtest_mtrap_routine=True -DTEST_CASE_1=True -DXLEN=32 -mabi=ilp32 ; spike --isa=rv32imfdc_Zba_Zbb_Zbc_Zbs +signature=/home/harris/cvw/tests/riscof/riscof_work/rv32i_m/C/src/cebreak-01.S/dut/DUT-spike.signature +signature-granularity=4 my.elf;

davidharrishmc, 2023-05-10:
I realized I was out of date on the repo.  After pulling the latest, I also am getting errors in arch_test

make[1]: Leaving directory '/home/harris/cvw/tests/riscof/riscof_work'[0m
[1;31m   ERROR[0m | [1;31m/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/env/arch_test.h: Assembler messages:
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/env/arch_test.h:760: Warning: RVMODEL_CLR_MSW_INT    not defined. Executing this will end test. Define an empty macro to suppress this warning
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/env/arch_test.h:764: Warning: RVMODEL_CLR_MTIMER_INT not defined. Executing this will end test. Define an empty macro to suppress this warning
...
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/env/arch_test.h:845: Error: unrecognized opcode `csrrw t3,CSR_XSCRATCH,t1', extension `zicsr' required
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cebreak-01.S:463:  Info: macro invoked from here
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cebreak-01.S:1635:   Info: macro invoked from here
/home/harris/cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/C/src/cebreak-01.S:19:    Info: macro invoked from here

pawks, 2023-05-15:
Looks like the `zicsr` extension is not being produced in the plugin. Can you check if the plugins are recent and upto date?

davidharrishmc, 2023-05-15:
Thank you! That solved it the zicsr issue.  We had old code mods.

The root issue is that riscof_spike.py gets customized.  Then other parts of riscof evolve, and riscof_spike.py becomes incompatible.  It is buried deeply enough and the changes were made by another member of the team, so it was hard to troubleshoot.

Allen addressed the RVMODEL warnings in
https://github.com/riscv-non-isa/riscv-arch-test/issues/336

closed 2023-05-15 by davidharrishmc