riscv-software-src/riscof #94: need to update command on how to run coverage and fix directory path where riscof looks for elf

dansmathers, 2023-08-13:
moving issue from https://github.com/riscv-software-src/riscv-config/issues/135

https://riscof.readthedocs.io/en/stable/coverage.html

The current directions say to run coverage:
riscof --verbose debug coverage --suite /path/to/suite --env /path/to/suite

this will generate an error. -c or --cgf_file is a required option.

However, even if -cgf_file is used, there is another error. riscof looks for the elf file in the wrong directory:

FileNotFoundError: [Errno 2] No such file or directory: '/riscof/riscof_work/rv32i_m/C/src/caddi-01.S/ref.elf'

but it looks like it should instead be looking in:
./rv32i_m/I/src/addi-01.S/ref/ref.elf

Thanks

I don't think the paths are wrong. Both would probably point to the same thing. All paths are resolved to be absolute paths before the actual execution of riscof starts. The former simply looks like the absolute path of the latter. I think the compilation was not successful. Can you share the entire log?

@dansmathers its better to move/track this issue in the [riscof repo](https://github.com/riscv-software-src/riscof) since it doesn't have anything to do with riscv-config.

dansmathers, 2023-08-13:
riscof coverage  --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env --cgf-file <path>/riscof/riscv-arch-test/coverage/rvi_c.cgf --cgf-file <path>/riscof/riscv-arch-test/coverage/rvi.cgf --cgf-file <path>/riscof/riscv-arch-test/coverage/rvi_fencei.cgf --cgf-file <path>/riscof/riscv-arch-test/coverage/rvi_m.cgf --cgf-file <path>/riscof/riscv-arch-test/coverage/rvi_priv.cgf | tee cov.log
    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.25.3 *******
    INFO | using riscv_isac version : 0.17.0
    INFO | using riscv_config version : 3.9.1
    INFO | Reading configuration from: <path>/riscof/config.ini
    INFO | Preparing Models
    INFO | Input-ISA file
    INFO | Loading input file: <path>/riscof/spike/spike_isa.yaml
    INFO | Load Schema <path>/.local/lib/python3.8/site-packages/riscv_config/schemas/schema_isa.yaml
    INFO | Processing Hart: hart0
    INFO | Initiating Validation
    INFO | No errors for Hart: 0 :)
    INFO |  Updating fields node for each CSR
    INFO | Dumping out Normalized Checked YAML: <path>/riscof/riscof_work/spike_isa_checked.yaml
    INFO | Input-Platform file
    INFO | Loading input file: <path>/riscof/spike/spike_platform.yaml
    INFO | Load Schema <path>/.local/lib/python3.8/site-packages/riscv_config/schemas/schema_platform.yaml
    INFO | Initiating Validation
    INFO | No Syntax errors in Input Platform Yaml. :)
    INFO | Dumping out Normalized Checked YAML: <path>/riscof/riscof_work/spike_platform_checked.yaml
    INFO | Generating database for suite: <path>/riscof/riscv-arch-test-20230726/riscv-test-suite
    INFO | Database File Generated: <path>/riscof/riscof_work/database.yaml
    INFO | Env path set to<path>/riscof/riscv-arch-test-20230726/riscv-test-suite/env
    INFO | Will collect Coverage using RISCV-ISAC
    INFO | CGF file(s) being used : ('<path>/riscof/riscv-arch-test-20230726/coverage/rvi_c.cgf', '<path>/riscof/riscv-arch-test-20230726/coverage/rvi.cgf', '<path>/riscof/riscv-arch-test-20230726/coverage/rvi_fencei.cgf', '<path>/riscof/riscv-arch-test-20230726/coverage/rvi_m.cgf', '<path>/riscof/riscv-arch-test-20230726/coverage/rvi_priv.cgf')
    INFO | Selecting Tests.
    INFO | Running Tests on Reference.
   ERROR |     INFO | ****** RISC-V ISA Coverage 0.17.0 *******
    INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd.
    INFO | All Rights Reserved.

Traceback (most recent call last):
  File "<path>/.local/bin/riscv_isac", line 8, in <module>
    sys.exit(cli())
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "<path>/.local/lib/python3.8/site-packages/riscv_isac/main.py", line 136, in coverage
    isac(output_file,elf,trace_file, window_size, expand_cgf(cgf_file,int(xlen),int(flen),log_redundant), parser_name, decoder_name, parser_path, decoder_path, detailed, test_label,
  File "<path>/.local/lib/python3.8/site-packages/riscv_isac/cgf_normalize.py", line 565, in expand_cgf
    cgf = utils.load_cgf(cgf_files)
  File "<path>/.local/lib/python3.8/site-packages/riscv_isac/utils.py", line 163, in load_cgf
    safe_yaml.dump(dict(safe_yaml.load(fp)),string_stream)
  File "<path>/.local/lib/python3.8/site-packages/ruamel/yaml/main.py", line 426, in load
    return constructor.get_single_data()
  File "<path>/.local/lib/python3.8/site-packages/ruamel/yaml/constructor.py", line 111, in get_single_data
    node = self.composer.get_single_node()
  File "_ruamel_yaml.pyx", line 706, in _ruamel_yaml.CParser.get_single_node
  File "_ruamel_yaml.pyx", line 724, in _ruamel_yaml.CParser._compose_document
  File "_ruamel_yaml.pyx", line 775, in _ruamel_yaml.CParser._compose_node
  File "_ruamel_yaml.pyx", line 889, in _ruamel_yaml.CParser._compose_mapping_node
  File "_ruamel_yaml.pyx", line 775, in _ruamel_yaml.CParser._compose_node
  File "_ruamel_yaml.pyx", line 889, in _ruamel_yaml.CParser._compose_mapping_node
  File "_ruamel_yaml.pyx", line 775, in _ruamel_yaml.CParser._compose_node
  File "_ruamel_yaml.pyx", line 889, in _ruamel_yaml.CParser._compose_mapping_node
  File "_ruamel_yaml.pyx", line 743, in _ruamel_yaml.CParser._compose_node
ruamel.yaml.composer.ComposerError: found undefined alias
  in "<file>", line 15, column 11
make: *** [<path>/riscof/riscof_work/Makefile.Reference-sail_c_simulator:5: TARGET0] Error 1
    INFO | Merging Coverage reports
Traceback (most recent call last):
  File "<path>/.local/bin/riscof", line 8, in <module>
    sys.exit(cli())
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "<path>/.local/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
[... cut ...]
    report, for_html, test_stats, coverpoints = framework.run_coverage(base, isa_file, platform_file,
  File "<path>/.local/lib/python3.8/site-packages/riscof/framework/main.py", line 126, in run_coverage
    'test_size': [str(entry) for entry in find_elf_size(elf)],
  File "<path>/.local/lib/python3.8/site-packages/riscof/framework/main.py", line 59, in find_elf_size
    with open(elf, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '<path>/riscof/riscof_work/rv32i_m/C/src/cadd-01.S/ref/ref.elf'

dansmathers, 2023-08-13:
@pawks Can you share your command you use to run coverage and I will try that?  I was guessing at the correct command because I didn't see a quickstart command that I could just run correctly?

Thanks,
Dan

pawks, 2023-08-13:
The error you are seeing is because you didnt include the dataset.cgf in the command line. It should be the first file in the list of cgf file arguments. The correct command should look something like the following:
```
riscof coverage --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env --cgf-file /riscof/riscv-arch-test/dataset.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_c.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_fencei.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_m.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_priv.cgf
```

neelgala, 2023-08-14:
@dansmathers as we noted in the last SIG call.. the docs for "test contributors" is missing and we will be fixing that asap in the coming weeks. It would be great if you could list out the missing docs you encountered in your struggle so we can accurately address those and others together.

Thanks for the inputs.

dansmathers, 2023-08-14:
Thanks.  that command helped.   I still see errors going by for some tests but the script ran.   I was trying to build everything from scratch again to check but it looks like the current github sail build is currently broken.   It does seem like it might be hard to do large unified pulls because there are 4 or 5 githubs that need to be kept in sync?

shriyan111, 2025-02-11:
> The error you are seeing is because you didnt include the dataset.cgf in the command line. It should be the first file in the list of cgf file arguments. The correct command should look something [...]

Hello, we followed the same command to generate the coverage report for config32e still we are facing the same error as discussed in the above thread and the error message is like,

riscof coverage --work-dir=./riscof_work --config=config32e.ini --suite=../../addins/riscv-arch-test/riscv-test-suite/ --env=../../addins/riscv-arch-test/riscv-test-suite/env --cgf-file=../../addins/riscv-arch-test/coverage/dataset.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_c.cgf  --cgf-file=../../addins/riscv-arch-test/coverage/rvi.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_fencei.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_m.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_priv.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rv32e.cgf
    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.25.3 *******
    INFO | using riscv_isac version : 0.18.0
    INFO | using riscv_config version : 3.18.3
    INFO | Reading configuration from: /home/satish/wally_core_putta_copy/cvw/tests/riscof/config32e.ini
    INFO | Preparing Models
    INFO | Input-ISA file
    INFO | ISACheck: Loading input file: /home/satish/wally_core_putta_copy/cvw/tests/riscof/spike/spike_rv32e_isa.yaml
    INFO | ISACheck: Load Schema /usr/local/lib/python3.10/dist-packages/riscv_config/schemas/schema_isa.yaml
    INFO | ISACheck: Processing Hart:0
    INFO | ISACheck: Initiating Validation for Hart:0
    INFO | ISACheck: No errors for Hart:0
    INFO | ISACheck:  Updating fields node for each CSR in Hart:0
    INFO | ISACheck: Dumping out Normalized Checked YAML: /home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/spike_rv32e_isa_checked.yaml
    INFO | Input-Platform file
    INFO | Loading input file: /home/satish/wally_core_putta_copy/cvw/tests/riscof/spike/spike_platform.yaml
    INFO | Load Schema /usr/local/lib/python3.10/dist-packages/riscv_config/schemas/schema_platform.yaml
    INFO | Initiating Validation
    INFO | No Syntax errors in Input Platform Yaml. :)
    INFO | Dumping out Normalized Checked YAML: /home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/spike_platform_checked.yaml
    INFO | Generating database for suite: /home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/riscv-test-suite
    INFO | Database File Generated: /home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/database.yaml
    INFO | Env path set to/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/riscv-test-suite/env
    INFO | Will collect Coverage using RISCV-ISAC
    INFO | CGF file(s) being used : ('/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/dataset.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_c.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_fencei.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_m.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_priv.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rv32e.cgf')
    INFO | Selecting Tests.
    INFO | Running Tests on Reference.
   ERROR | make[1]: Entering directory '/home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work'
make[1]: Leaving directory '/home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work'
   ERROR |     INFO | ****** RISC-V ISA Coverage 0.18.0 *******
    INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd.
    INFO | All Rights Reserved.
    INFO | ****** RISC-V ISA Coverage 0.18.0 *******
    INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd.
    INFO | All Rights Reserved.
    INFO | ****** RISC-V ISA Coverage 0.18.0 *******
    INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd.
    INFO | All Rights Reserved.
Traceback (most recent call last):
  File "//usr/local/bin/riscv_isac", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
Traceback (most recent call last):
  File "//usr/local/bin/riscv_isac", line 8, in <module>
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/riscv_isac/main.py", line 136, in coverage
    isac(output_file,elf,trace_file, window_size, expand_cgf(cgf_file,int(xlen),int(flen),log_redundant), parser_name, decoder_name, parser_path, decoder_path, detailed, test_label,
  File "/usr/local/lib/python3.10/dist-packages/riscv_isac/cgf_normalize.py", line 565, in expand_cgf
[... cut ...]
  File "/usr/local/lib/python3.10/dist-packages/riscof/framework/main.py", line 126, in run_coverage
    'test_size': [str(entry) for entry in find_elf_size(elf)],
  File "/usr/local/lib/python3.10/dist-packages/riscof/framework/main.py", line 59, in find_elf_size
    with open(elf, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/rv32e_m/C/src/caddi16sp-01.S/ref.elf'
make: *** [Makefile:27: coverage] Error 1

is there any other command we should include to  overcome this error and get correct coverage report?

Sriguru840, 2025-02-12:
> > The error you are seeing is because you didnt include the dataset.cgf in the command line. It should be the first file in the list of cgf file arguments. The correct command should look something like the following:
> > ```
> > riscof coverage --config=config.ini --suite=riscv-arch-test/riscv-test-suite/ --env=riscv-arch-test/riscv-test-suite/env --cgf-file /riscof/riscv-arch-test/dataset.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_c.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_fencei.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_m.cgf --cgf-file /riscof/riscv-arch-test/coverage/rvi_priv.cgf
> > ```
> 
> Hello, we followed the same command to generate the coverage report for config32e still we are facing the same error as discussed in the above thread and the error message is like,
> 
> riscof coverage --work-dir=./riscof_work --config=config32e.ini --suite=../../addins/riscv-arch-test/riscv-test-suite/ --env=../../addins/riscv-arch-test/riscv-test-suite/env --cgf-file=../../addins/riscv-arch-test/coverage/dataset.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_c.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_fencei.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_m.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rvi_priv.cgf --cgf-file=../../addins/riscv-arch-test/coverage/rv32e.cgf INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.25.3 ******* INFO | using riscv_isac version : 0.18.0 INFO | using riscv_config version : 3.18.3 INFO | Reading configuration from: /home/satish/wally_core_putta_copy/cvw/tests/riscof/config32e.ini INFO | Preparing Models INFO | Input-ISA file INFO | ISACheck: Loading input file: /home/satish/wally_core_putta_copy/cvw/tests/riscof/spike/spike_rv32e_isa.yaml INFO | ISACheck: Load Schema /usr/local/lib/python3.10/dist-packages/riscv_config/schemas/schema_isa.yaml INFO | ISACheck: Processing Hart:0 INFO | ISACheck: Initiating Validation for Hart:0 INFO | ISACheck: No errors for Hart:0 INFO | ISACheck: Updating fields node for each CSR in Hart:0 INFO | ISACheck: Dumping out Normalized Checked YAML: /home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/spike_rv32e_isa_checked.yaml INFO | Input-Platform file INFO | Loading input file: /home/satish/wally_core_putta_copy/cvw/tests/riscof/spike/spike_platform.yaml INFO | Load Schema /usr/local/lib/python3.10/dist-packages/riscv_config/schemas/schema_platform.yaml INFO | Initiating Validation INFO | No Syntax errors in Input Platform Yaml. :) INFO | Dumping out Normalized Checked YAML: /home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/spike_platform_checked.yaml INFO | Generating database for suite: /home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/riscv-test-suite INFO | Database File Generated: /home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/database.yaml INFO | Env path set to/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/riscv-test-suite/env INFO | Will collect Coverage using RISCV-ISAC INFO | CGF file(s) being used : ('/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/dataset.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_c.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_fencei.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_m.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rvi_priv.cgf', '/home/satish/wally_core_putta_copy/cvw/addins/riscv-arch-test/coverage/rv32e.cgf') INFO | Selecting Tests. INFO | Running Tests on Reference. ERROR | make[1]: Entering directory '/home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work' make[1]: Leaving directory '/home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work' ERROR | INFO | ****** RISC-V ISA Coverage 0.18.0 ******* INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd. INFO | All Rights Reserved. INFO | ****** RISC-V ISA Coverage 0.18.0 ******* INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd. INFO | All Rights Reserved. INFO | ****** RISC-V ISA Coverage 0.18.0 ******* INFO | Copyright (c) 2020, InCore Semiconductors Pvt. Ltd. INFO | All Rights Reserved. Traceback (most recent call last): File "//usr/local/bin/riscv_isac", line 8, in sys.exit(cli()) File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in **call** return self.main(*args, **kwargs) File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main rv = self.invoke(ctx) File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke Traceback (most recent call last): File "//usr/local/bin/riscv_isac", line 8, in return _process_result(sub_ctx.command.invoke(sub_ctx)) File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke sys.exit(cli()) File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in **call** return ctx.invoke(self.callback, **ctx.params) File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke return self.main(*args, **kwargs) File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main return __callback(*args, **kwargs) File "/usr/local/lib/python3.10/dist-packages/riscv_isac/main.py", line 136, in coverage isac(output_file,e [... cut ...] work/main.py", line 59, in find_elf_size with open(elf, 'rb') as f: FileNotFoundError: [Errno 2] No such file or directory: '/home/satish/wally_core_putta_copy/cvw/tests/riscof/riscof_work/rv32e_m/C/src/caddi16sp-01.S/ref.elf' make: *** [Makefile:27: coverage] Error 1
> 
> is there any other command we should include to overcome this error and get correct coverage report?

Hello,
I am also facing the same issue.
I am not getting what command is missed out. Can anyone help me out with this issue!