riscv-software-src/riscof #122: riscv_isac dependency not updated

5hayanB, 2024-07-29:
Cloned the `riscof` framework via the command which went smoothly:
```zsh
pip3 install git+https://github.com/riscv/riscof.git
```

However, running `riscof` via the command:
```zsh
riscof --verbose info arch-test --clone
```
I am greeted with the following error:
```zsh
File "$HOME/.venv/bin/riscof", line 5, in <module>
    from riscof.cli import cli
  File "$HOME/.venv/lib/python3.12/site-packages/riscof/cli.py", line 18, in <module>
    import riscof.framework.main as framework
  File "$HOME/.venv/lib/python3.12/site-packages/riscof/framework/main.py", line 11, in <module>
    from riscv_isac.isac import preprocessing
ImportError: cannot import name 'preprocessing' from 'riscv_isac.isac' ($HOME/.venv/lib/python3.12/site-packages/riscv_isac/isac.py)
```

On debugging the problem, I found out that the recent commit [`dddbc52`](https://github.com/riscv-software-src/riscof/commit/dddbc52b1cabd6b4571b185bc60f910eef89e6c9) added import statements from [`isac.py`](https://github.com/riscv-software-src/riscv-isac/blob/dev/riscv_isac/isac.py) of the `dev` branch. But the `riscv_isac` dependency is not updated in [`requirements.txt`](https://github.com/riscv-software-src/riscof/blob/dev/riscof/requirements.txt). Reverting to commit [`d38859f`](https://github.com/riscv-software-src/riscof/commit/d38859f85fe407bcacddd2efcd355ada4683aee4) fixes the problem.

stnolting-ims, 2024-07-30:
I can confirm this bug (https://github.com/stnolting/neorv32-riscof/actions/runs/10156350492/job/28086317621#step:7:43).

> Reverting to commit [d38859f](https://github.com/riscv-software-src/riscof/commit/d38859f85fe407bcacddd2efcd355ada4683aee4) fixes the problem.

This seems to fix this (as a temporary workaround).

stnolting, 2024-08-01:
It this going to be fixed? Shall we provide a PR to address this?

5hayanB, 2025-01-08:
I have opened pr #123 as a temporary fix.

jeras, 2025-03-15:
Having the same issue, as per above comments I used the following CLI to install `d38859f` (also tried `dev` but it had the same issue):
```
pip3 install git+https://github.com/riscv/riscof.git@d38859f
```