riscv-software-src/riscof #128: Import preprocessing error

itsubaidrehman, 2024-12-14:
I am trying to install RISCoF framework but getting preprocessing error.

File "/home/lpt-10xe/riscof/riscof/cli.py", line 18, in <module>
    import riscof.framework.main as framework
  File "/home/lpt-10xe/riscof/riscof/framework/main.py", line 11, in <module>
    from riscv_isac.isac import preprocessing
ImportError: cannot import name 'preprocessing' from 'riscv_isac.isac' (/home/lpt-10xe/.local/lib/python3.10/site-packages/riscv_isac/isac.py)

jordancarlin, 2024-12-15:
This is caused by #122. For now the workaround is to install a newer version of riscv-isac manually. The instructions in that issue are no longer up to date because riscv-isac has been moved to the riscv-arch-test repo. The following command should work:

```
pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
```

jordancarlin, 2024-12-15:
@UmerShahidengr It seems like there won't be further packaged pip releases of riscv-isac now that it has been directly incorporated into riscv-arch-test. That leaves the default install of riscof unusable in its current state. Is there a reason not to incorporate riscof into the riscv-arch-test repo as well to avoid these issues?

itsubaidrehman, 2024-12-15:
@jordancarlin @UmerShahidengr 
To avoid the preprocessing impor error, Installing RISCoF using pip3 worked for me.
pip3 install riscof

jordancarlin, 2024-12-15:
@itsubaidrehman Unfortunately that installs an older version from January 2023. At a minimum I believe that version is missing features needed for testing the privileged architecture.

itsubaidrehman, 2024-12-15:
@jordancarlin I tried to install using 
pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
and now i am getting the same above mentioned pre-processing error. Beside that now pip3 install riscof is also unable to install riscof.

jordancarlin, 2024-12-15:
Maybe try uninstalling riscof and riscv-isac first? Maybe also try removing everything from the pip cache? `python -m pip cache purge`

The following two commands work to install riscv-isac and riscof in a functional state for me.
```
pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
pip install git+https://github.com/riscv/riscof
```

The other option I can think of trying is to replace the current riscv-isac line in the requirements.txt file with `riscv-isac @ git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac` instead.

itsubaidrehman, 2024-12-15:
@jordancarlin I have checkout to d38859f commit and then installed riscof and it worked. Now i did not get import preprocessing error.

jordancarlin, 2024-12-15:
That makes sense. It was commit https://github.com/riscv-software-src/riscof/commit/dddbc52b1cabd6b4571b185bc60f910eef89e6c9 that introduced the feature that is dependent on the newer version of riscv-isac. Some of those features will be needed for privileged tests, but if it works fine for now without it then I guess maybe that's good enough.

itsubaidrehman, 2024-12-15:
but still, if some one has to use those features, what could be the possible solutions for avoid this error?

jordancarlin, 2024-12-15:
The only solutions I'm aware of are the two I listed above. Not sure why neither worked for you. 

Ideally the longer term solution would be to merge riscof into riscv-arch-test so that they can't get out of sync and don't have weird dependencies.

itsubaidrehman, 2024-12-23:
@jordancarlin I have installed riscof with these two commands
**pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
pip install git+https://github.com/riscv/riscof**
but now coverage command is working fine for M extension but for I extension i am getting these errors

**ERROR | Covergroup(s) for ('CHK_M_INTR_PRIORITY',) not found
make: *** [/home/lpt-10xe/WORK/riscof_work/Makefile.Reference-sail_c_simulator:9: TARGET1] Error 1**
    INFO | Merging Coverage reports
Traceback (most recent call last):
  File "/home/lpt-10xe/.local/bin/riscof", line 8, in <module>
    sys.exit(cli())
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/riscof/cli.py", line 406, in coverage
    report, for_html, test_stats, coverpoints = framework.run_coverage(base, isa_file, platform_file,
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/riscof/framework/main.py", line 129, in run_coverage
    'test_size': [str(entry) for entry in find_elf_size(elf)],
  File "/home/lpt-10xe/.local/lib/python3.10/site-packages/riscof/framework/main.py", line 61, in find_elf_size
    with open(elf, 'rb') as f:
**FileNotFoundError: [Errno 2] No such file or directory: '/home/lpt-10xe/WORK/riscof_work/src/addi-01.S/ref.elf'**

jordancarlin, 2024-12-23:
I've spent a while working on installation related issues but am not very familiar with actually using riscof for coverage, so I'm not sure that I can be much help at this point.

zaun, 2025-01-03:
Just chiming in, Im running into this issue as well:

```
╰─$ pipx install --editable . --force                                                                                                   
--python is ignored when --force is passed. If you want to reinstall riscv_ctg with
/Users/justinzaun/local/brew/opt/python@3.12/libexec/bin/python, run `pipx reinstall . --python
/Users/justinzaun/local/brew/opt/python@3.12/libexec/bin/python` instead.
Installing to existing venv 'riscv_ctg'
  installed package riscv_ctg 0.12.2, installed using Python 3.12.6
  These apps are now globally available
    - riscv_ctg
done! ✨ 🌟 ✨
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/riscv-ctg ‹dev› 
╰─$ cd ..
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ cd riscv-isac   
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/riscv-isac ‹dev› 
╰─$ pipx install --editable . --force
--python is ignored when --force is passed. If you want to reinstall riscv_isac with
/Users/justinzaun/local/brew/opt/python@3.12/libexec/bin/python, run `pipx reinstall . --python
/Users/justinzaun/local/brew/opt/python@3.12/libexec/bin/python` instead.
Installing to existing venv 'riscv_isac'
  installed package riscv_isac 0.18.0, installed using Python 3.12.6
  These apps are now globally available
    - riscv_isac
done! ✨ 🌟 ✨
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/riscv-isac ‹dev› 
╰─$ cd ..                            
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ riscof --help
Traceback (most recent call last):
  File "/Users/justinzaun/local/bin/riscof", line 5, in <module>
    from riscof.cli import cli
  File "/Users/justinzaun/local/venvs/riscof/lib/python3.12/site-packages/riscof/cli.py", line 18, in <module>
    import riscof.framework.main as framework
  File "/Users/justinzaun/local/venvs/riscof/lib/python3.12/site-packages/riscof/framework/main.py", line 11, in <module>
    from riscv_isac.isac import preprocessing
ImportError: cannot import name 'preprocessing' from 'riscv_isac.isac' (/Users/justinzaun/local/venvs/riscof/lib/python3.12/site-packages/riscv_isac/isac.py)
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$                                                                                                                                       1 ↵

```

itsubaidrehman, 2025-01-03:
@zaun hopefully these two commands will install the RISCoF framwork setup without these error.
pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
pip install git+https://github.com/riscv/riscof

zaun, 2025-01-03:
@itsubaidrehman Unfortunately no.

```
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ pipx uninstall riscv-isac                                                                                                             2 ↵
uninstalled riscv_isac! ✨ 🌟 ✨
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ pipx uninstall riscof
uninstalled riscof! ✨ 🌟 ✨
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ pipx install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
  installed package riscv_isac 0.18.0, installed using Python 3.12.6
  These apps are now globally available
    - riscv_isac
done! ✨ 🌟 ✨
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ pipx install git+https://github.com/riscv/riscof
  installed package riscof 1.25.3, installed using Python 3.12.6
  These apps are now globally available
    - riscof
done! ✨ 🌟 ✨
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ riscof
Traceback (most recent call last):
  File "/Users/justinzaun/local/bin/riscof", line 5, in <module>
    from riscof.cli import cli
  File "/Users/justinzaun/local/venvs/riscof/lib/python3.12/site-packages/riscof/cli.py", line 18, in <module>
    import riscof.framework.main as framework
  File "/Users/justinzaun/local/venvs/riscof/lib/python3.12/site-packages/riscof/framework/main.py", line 11, in <module>
    from riscv_isac.isac import preprocessing
ImportError: cannot import name 'preprocessing' from 'riscv_isac.isac' (/Users/justinzaun/local/venvs/riscof/lib/python3.12/site-packages/riscv_isac/isac.py)
```

jordancarlin, 2025-01-03:
@itsubaidrehman Can you try installing using pip inside of a single virtual environment instead of pipx to see if it is an issue with the way pipx is handling it?

```
python -m venv test
source test/bin/activate
pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
pip install git+https://github.com/riscv/riscof
```

zaun, 2025-01-03:
So, yeah that works. I'm not dont know python or pip/pipx well enough to know why it worked and pipx didn't though. Maybe is something with being on macOS or python/pipx being from brew? This is the full log:

```
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev› 
╰─$ python -m venv test                                                                                                                   1 ↵
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev●› 
╰─$ source test/bin/activate
(test) ╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test ‹dev●› 
╰─$ pip install git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
Collecting git+https://github.com/riscv-non-isa/riscv-arch-test/#subdirectory=riscv-isac
  Cloning https://github.com/riscv-non-isa/riscv-arch-test/ to /private/var/folders/2n/9d_8f0dd7tj6bwwx3tmxrgn80000gn/T/pip-req-build-5pw5hw_a
  Running command git clone --filter=blob:none --quiet https://github.com/riscv-non-isa/riscv-arch-test/ /private/var/folders/2n/9d_8f0dd7tj6bwwx3tmxrgn80000gn/T/pip-req-build-5pw5hw_a
  Resolved https://github.com/riscv-non-isa/riscv-arch-test/ to commit 1b1fb26da37a2a426dd1f5260b0f08ff4e321f0e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting alabaster>=0.7.12 (from riscv_isac==0.18.0)
  Using cached alabaster-1.0.0-py3-none-any.whl.metadata (2.8 kB)
Collecting Babel>=2.7.0 (from riscv_isac==0.18.0)
  Using cached babel-2.16.0-py3-none-any.whl.metadata (1.5 kB)
Collecting Cerberus>=1.3.1 (from riscv_isac==0.18.0)
  Using cached Cerberus-1.3.7-py3-none-any.whl.metadata (5.5 kB)
Collecting certifi>=2019.6.16 (from riscv_isac==0.18.0)
  Using cached certifi-2024.12.14-py3-none-any.whl.metadata (2.3 kB)
Collecting chardet>=3.0.4 (from riscv_isac==0.18.0)
  Using cached chardet-5.2.0-py3-none-any.whl.metadata (3.4 kB)
Collecting click (from riscv_isac==0.18.0)
  Using cached click-8.1.8-py3-none-any.whl.metadata (2.3 kB)
Collecting colorlog (from riscv_isac==0.18.0)
  Using cached colorlog-6.9.0-py3-none-any.whl.metadata (10 kB)
Collecting doc8>=0.8.0 (from riscv_isac==0.18.0)
  Using cached doc8-1.1.2-py3-none-any.whl.metadata (8.3 kB)
Collecting docutils>=0.14 (from riscv_isac==0.18.0)
  Using cached docutils-0.21.2-py3-none-any.whl.metadata (2.8 kB)
Collecting gitdb2>=2.0.5 (from riscv_isac==0.18.0)
  Using cached gitdb2-4.0.2-py3-none-any.whl.metadata (383 bytes)
Collecting gitpython (from riscv_isac==0.18.0)
  Using cached GitPython-3.1.44-py3-none-any.whl.metadata (13 kB)
Collecting idna>=2.8 (from riscv_isac==0.18.0)
  Using cached idna-3.10-py3-none-any.whl.metadata (10 kB)
Collecting imagesize>=1.1.0 (from riscv_isac==0.18.0)
  Using cached imagesize-1.4.1-py2.py3-none-any.whl.metadata (1.5 kB)
Collecting Jinja2 (from riscv_isac==0.18.0)
  Using cached jinja2-3.1.5-py3-none-any.whl.metadata (2.6 kB)
Collecting m2r2>=0.2.7 (from riscv_isac==0.18.0)
  Using cached m2r2-0.3.3.post2-py3-none-any.whl.metadata (7.1 kB)
Collecting MarkupSafe>=1.1.1 (from riscv_isac==0.18.0)
  Using cached MarkupSafe-3.0.2-cp312-cp312-macosx_11_0_arm64.whl.metadata (4.0 kB)
Collecting mistune>=0.8.4 (from riscv_isac==0.18.0)
  Using cached mistune-3.1.0-py3-none-any.whl.metadata (1.7 kB)
Collecting ordered-set>=4.1.0 (from riscv_isac==0.18.0)
  Using cached ordered_set-4.1.0-py3-none-any.whl.metadata (5.3 kB)
Collecting oyaml>=0.9 (from riscv_isac==0.18.0)
  Using cached oyaml-1.0-py2.py3-none-any.whl.metadata (1.2 kB)
Collecting packaging>=19.0 (from riscv_isac==0.18.0)
  Using cached packaging-24.2-py3-none-any.whl.metadata (3.2 kB)
Collecting pbr>=5.3.1 (from riscv_isac==0.18.0)
  Using cached pbr-6.1.0-py2.py3-none-any.whl.metadata (3.4 kB)
Collecting pluggy (from riscv_isac==0.18.0)
  Using cached pluggy-1.5.0-py3-none-any.whl.metadata (4.8 kB)
Collecting pyelftools>=0.26 (from riscv_isac==0.18.0)
  Using cached pyelftools-0.31-py3-none-any.whl.metadata (381 bytes)
Collecting Pygments>=2.4.2 (from riscv_isac==0.18.0)
  Using cached pygments-2.18.0-py3-none-any.whl.metadata (2.5 kB)
Collecting pyparsing>=2.4.0 (from riscv_isac==0.18.0)
  Using cached pyparsing-3.2.1-py3-none-any.whl.metadata (5.0 kB)
Collecting pytablewriter (from riscv_isac==0.18.0)
  Using cached pytablewriter-1.2.1-py3-none-any.whl.metadata (38 kB)
Collecting pytest (from riscv_isac==0.18.0)
  Using cached pytest-8.3.4-py3-none-any.whl.metadata (7.5 kB)
[... 333 lines cut ...]
  arch-test     Setup and maintenance for Architectural TestSuite.
  coverage      Run the tests on DUT and reference and compare signatures
  gendb         Generate Database for the Suite.
  run           Run the tests on DUT and reference and compare signatures
  setup         Initiate Setup for riscof.
  testlist      Generate the test list for the given DUT and suite.
  validateyaml  Validate the Input YAMLs using riscv-config.
```
  
  If it helps, these are my normal versions:
  
```
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/test ‹dev●› 
╰─$ python --version
Python 3.12.6
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/test ‹dev●› 
╰─$ which python    
/Users/justinzaun/local/bin/python
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/test ‹dev●› 
╰─$ pipx --version
1.5.0
╭─justinzaun@Hogsmeade ~/Development/riscv-arch-test/test ‹dev●› 
╰─$ which pipx
/Users/justinzaun/local/brew/bin/pipx
```

jordancarlin, 2025-01-03:
I think pipx installs everything in its own dedicated virtual environment, so when using pipx the newer version of riscv-isac is probably not visible to riscof and it falls back to using the older one that is published to PyPi. 

There is probably a way around that, but I haven't used pipx so am not familiar with what that would be.

3Mochi, 2025-11-23:
Hi, I'm still having the same issue. I followed the instructions in #147, and installed the latest versions of `riscof`, `riscv-config` and `riscv-isac`. But running `riscof` in the command line produced this.

```
Traceback (most recent call last):
  File "/home/my_username/workspace/bin/./riscof", line 5, in <module>
    from riscof.cli import cli
  File "/home/my_username/workspace/lib/python3.12/site-packages/riscof/cli.py", line 18, in <module>
    import riscof.framework.main as framework
  File "/home/my_username/workspace/lib/python3.12/site-packages/riscof/framework/main.py", line 11, in <module>
    from riscv_isac.isac import preprocessing
ImportError: cannot import name 'preprocessing' from 'riscv_isac.isac' (/home/my_username/workspace/lib/python3.12/site-packages/riscv_isac/isac.py)
```

mmichilot, 2025-11-23:
I notice that install order does matter since if you install `riscof` first, it will still somehow use the older `riscv-isac` from PyPI, even if you installed `riscv-isac` from GitHub after. What worked for me was uninstalling `riscv-isac` via pip, then installing it again from GitHub. I would also recommend using a python virtual environment or miniconda so that you have a clean python environment to work with.