riscv-software-src/riscof #132: Reference plugin loops infinitely

GenericNerdyUsername, 2024-12-26:
Running the tests works fine for the DUT (ive tried both spike and serv), but sail loops infinitely, with the following error repeated on stdout:
```
within_phys_mem: 0x00000000 not within phys-mem:
  plat_rom_base: 0x00001000
  plat_rom_size: 0x00001000
  plat_ram_base: 0x80000000
  plat_ram_size: 0x04000000
trapping from M to M to handle fetch-access-fault
handling exc#0x01 at priv M with tval 0x00000000
CSR mstatus <- 0x00001800
```

I am on nixos, using riscof 1.25.3, riscv_isac 0.18.0, riscv_config 3.18.3, toolchain version 2024.12.16 and sail version 0.5

jordancarlin, 2024-12-28:
When you say you are using sail version 0.5 do you mean this [tagged version from 2020](https://github.com/riscv/sail-riscv/releases/tag/0.5)? If so, I'd recommend pulling the latest master branch and trying that. There have been a lot of changes to the model since then.

GenericNerdyUsername, 2024-12-28:
Thanks, that got it to finish. There are some test fails, but I think that's from spike_parallel being three years old, and I'm only using that to test my setup, not the actual processor I'm interested in, so I don't mind that

closed 2024-12-28 by GenericNerdyUsername

jordancarlin, 2024-12-28:
There shouldn’t be any failing tests between Sail and spike. What do you mean by spike_parallel? Are you using the tandem verification linking between Sail and Spike? That shouldn’t be necessary (and as you mentioned hasn’t been updated in years). You should be able to build the sail model normally just using make.