riscv-software-src/riscof #140: Unknown field error

txwhqq, 2025-04-28:
I try to execute the riscof validateyaml but get the errors showed as bellow:

calvin@ubuntu:~/distrobox/work/spike_test$ riscof validateyaml --config=config.ini
    INFO | ****** RISCOF: RISC-V Architectural Test Framework 1.25.3 *******
    INFO | using riscv_isac version : 0.18.0
    INFO | using riscv_config version : 3.18.3
    INFO | Reading configuration from: /home/calvin/distrobox/work/spike_test/config.ini
    INFO | Preparing Models
    INFO | Input-ISA file
    INFO | ISACheck: Loading input file: /home/calvin/distrobox/work/spike_test/spike/spike_isa.yaml
    INFO | ISACheck: Load Schema /home/calvin/.local/lib/python3.8/site-packages/riscv_config/schemas/schema_isa.yaml
    INFO | ISACheck: Processing Hart:0
    INFO | ISACheck: Initiating Validation for Hart:0
   ERROR | ISACheck: Error in /home/calvin/distrobox/work/spike_test/spike/spike_isa.yaml.
   ERROR |  PMP:
   ERROR |   - unknown field

codygunton, 2025-04-28:
I haven't understood what exactly this means, but try removing the PMP field from your spike_isa.yaml 😅 I managed to get to a fill passing suite of tests spike <> sail model this way

nibrunieAtSi5, 2025-10-04:
I believe this is actually an error raised by **riscv-config**
https://github.com/riscv-software-src/riscv-config/blob/54171f205be802f9f8e0b1cf4156a6cc826fb467/riscv_config/checker.py#L1843-L1851

I think it comes from the fact that the version of riscv-config and riscof are not aligned.

I was able to get around this error by forcing an install of **riscv-config** from the dev branch through:
```
pip3 install git+https://github.com/riscv-software-src/riscv-config.
git@dev
```

tom-gall, 2025-11-04:
Commenting here as well pip3 install git+https://github.com/riscv-software-src/riscv-config.git@dev fixed it for me. (I'm using a venv FWIW)