riscv-software-src/riscof #144: Pypi package for riscof doesn’t include latest commits

nadime15, 2025-07-28:
A colleague of mind send me the following screenshot:

[image]

Basically with the following Error Message:

```
ISACheck: Error in /...
PMP:
 - unknown field
```

I remember I ran into this at some point a few months ago as well. And I think its because commit be84132874963e001c14d846a140a3edd9c9d48f as not part of the pypi package and the package needs to be updated with a new version number.

llTurtle22ll, 2025-09-22:
Have you found any workaround/fix for this issue?
I've been stuck with this same problem for a while now.

For more context, I tried deleting this field from the yaml file, with no success, as this raises an error more down the execution where it tries to search for the PMP field in the yaml. Basically this is telling that the PMP field has to be there, but it cannot at the same time.

Also tried to compile from source, with no success as well, giving me some python dependency errors.

nibrunieAtSi5, 2025-10-04:
I think you can circumvent the issue by forcing the install of the package from source from the `dev` branch:
```
pip3 install git+https://github.com/riscv-software-src/riscv-config.git@dev
```

nibrunieAtSi5, 2025-10-04:
It looks like the error is simulator to https://github.com/riscv-software-src/riscof/issues/140
```
ISACheck: Error in /...
PMP:
 - unknown field
```